grzegorz914/homebridge-enphase-envoy #189: Envoy power mis reported as 100% when 0 production

steve-lad, 2025-03-21:
At night, the envoy brightness is set at 100% when there is actually 0 production. As soon as there is nominal production, the output is correct.

grzegorz914, 2025-03-21:
Try 9.9.9-beta and let me know

steve-lad, 2025-03-21:
This may or maybe not interesting from debug output

```
 "production": [
    {
      "type": "inverters",
      "activeCount": 12,
      "readingTime": 1742581830,
      "wNow": 0,
      "whLifetime": 658031
    },
    {
      "type": "eim",
      "activeCount": 1,
      "measurementType": "production",
      "readingTime": 1742582672,
      "wNow": 1.305,
      "whLifetime": 651222.546,
      "varhLeadLifetime": 235383.85,
      "varhLagLifetime": 115988.921,
      "vahLifetime": 847684.432,
      "rmsCurrent": 0.865,
      "rmsVoltage": 722.61,
      "reactPwr": 193.484,
      "apprntPwr": 208.274,
      "pwrFactor": 0,
      "whToday": 16596.546,
      "whLastSevenDays": 146825.546,
      "vahToday": 16640.432,
      "varhLeadToday": 4953.85,
      "varhLagToday": 1885.921,
      "lines": [
        {
          "wNow": 1.386,
          "whLifetime": 441021.329,
          "varhLeadLifetime": 160353.751,
          "varhLagLifetime": 85661.449,
          "vahLifetime": 571166.481,
          "rmsCurrent": 0.299,
          "rmsVoltage": 241.807,
          "reactPwr": 67.364,
          "apprntPwr": 72.099,
          "pwrFactor": 0,
          "whToday": 6247.329,
          "whLastSevenDays": 49667.329,
          "vahToday": 5665.481,
          "varhLeadToday": 1724.751,
          "varhLagToday": 605.449
        },
        {
          "wNow": 0.549,
          "whLifetime": 148556.519,
          "varhLeadLifetime": 53027.937,
          "varhLagLifetime": 21155.139,
          "vahLifetime": 175101.189,
          "rmsCurrent": 0.245,
          "rmsVoltage": 240.934,
          "reactPwr": 59.049,
          "apprntPwr": 59.049,
          "pwrFactor": 0,
          "whToday": 4685.519,
          "whLastSevenDays": 48655.519,
          "vahToday": 5479.189,
          "varhLeadToday": 1663.937,
          "varhLagToday": 638.139
        },
        {
          "wNow": -0.63,
          "whLifetime": 61644.697,
          "varhLeadLifetime": 22002.163,
          "varhLagLifetime": 9172.333,
          "vahLifetime": 101416.762,
          "rmsCurrent": 0.321,
          "rmsVoltage": 239.869,
          "reactPwr": 67.071,
          "apprntPwr": 77.125,
          "pwrFactor": 0,
          "whToday": 5663.697,
          "whLastSevenDays": 48502.697,
          "vahToday": 5495.762,
          "varhLeadToday": 1565.163,
          "varhLagToday": 642.333
        }
      ]
    }
  ],
```

```
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Production power state: true.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Production power level: 0 %.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Production power peak detected: true.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Production energy state: true.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Consumption Total power state: true.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Consumption Total energy state: true.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Consumption Net power state: true.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Consumption Net energy state: false.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Requesting production all.
[3/21/2025, 7:44:32 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Production all:  {
```

grzegorz914, 2025-03-21:
Here is off

[image]

steve-lad, 2025-03-21:
Does that not mean that Homekit requires a different brightness level (or state) if 0, ie off?

grzegorz914, 2025-03-21:
I have some idea, I will look into this.

steve-lad, 2025-03-21:
This is a capture from Homebridge, so the issue is not HomeKit

[image]

grzegorz914, 2025-03-21:
Try beta 1

steve-lad, 2025-03-21:
Looks like you have it:

[image]

Once gains, well done and thanks

grzegorz914, 2025-03-21:
v9.9.9 released

closed 2025-03-21 by grzegorz914