grzegorz914/homebridge-enphase-envoy #191: Suggestion to improve encharges display in Home

steve-lad, 2025-03-22:
as you are aware, encharges have a minimum SoC, typically 10%, but user/installer definable.

Suggestion: add a field for this minimum SoC to Encharges. When the battery is below this minimum, show 0 (off). Above this minimum, show the actual SoC as at present.

This would give a good quick glance view of their status in the home summary - off means that they are depleted and any other state shows them as available.

This is eye candy, so should not be a priority, assuming that you agree.

grzegorz914, 2025-03-22:
try 9.9.11-beta, please configure the light bulb accessory in plugin config first

[image]

steve-lad, 2025-03-22:
`Device: 192.168.30.150 PV Array, Error: Start error: Error: Update ensemble inventory error: ReferenceError: scaleValue is not defined, trying again.`

grzegorz914, 2025-03-22:
fixed

steve-lad, 2025-03-22:
Same in Beta 5

grzegorz914, 2025-03-22:
beta.6

steve-lad, 2025-03-22:
` Device: 192.168.30.150 PV Array, Error: Start error: Error: Update ensemble inventory error: ReferenceError: i is not defined, trying again.`

grzegorz914, 2025-03-22:
fixed b.7

steve-lad, 2025-03-22:
Fixed the start error, but there seems to be an implementation error - SOC is set at 10% and the battery is at 70%, so should be showing charge level

[image]

grzegorz914, 2025-03-22:
try b9

steve-lad, 2025-03-22:
Not available (Beta 9)

steve-lad, 2025-03-22:
Still off. App running, but this shows in logs

`This plugin generated a warning from the characteristic 'Brightness': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.`

grzegorz914, 2025-03-22:
fixed b10

steve-lad, 2025-03-22:
You have got it. Well done. Will confirm tomorrow when battery has discharged. Have a good evening and thanks

grzegorz914, 2025-03-22:
OK, i will publish release tomorrow after confirm

grzegorz914, 2025-03-22:
I can not test it itself because do not have encharges installed in my pv system

steve-lad, 2025-03-22:
Not an issue. Very happy to help with any testing or if you want to bounce ideas - just give me a shout

grzegorz914, 2025-03-22:
b11 added `2` - Fan, `3` - Humidity Sensor, `4` - Carbon Monoxide Sensor as a accessory type

grzegorz914, 2025-03-22:
9.10.0-beta added possibility to select accessory type for PV and Encharges

1 - Light Bulb, 2 - Fan, 3 - Humidity Sensor, 4 - Carbon Monoxide Sensor

steve-lad, 2025-03-23:
9.9.11 beta 10 Works perfectly. A great improvement giving instant visual feedback of state. Thanks

9.10.0-beta 1 Fails to load. I will be away for a week now, so will try latest when I return.

```
TypeError: Cannot read properties of undefined (reading 'displayType')
    at HomebridgeAPI.<anonymous> (file:///usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:32:54)
    at HomebridgeAPI.emit (node:events:518:28)
    at HomebridgeAPI.signalFinished (/usr/local/lib/node_modules/homebridge/src/api.ts:275:10)
    at ChildBridgeFork.startBridge (/usr/local/lib/node_modules/homebridge/src/childBridgeFork.ts:19
```

grzegorz914, 2025-03-23:
Update config UI

grzegorz914, 2025-03-23:
Try beta 7

grzegorz914, 2025-03-23:
OK, 9.10.0 released, I think all found bugs are fixed.

steve-lad, 2025-03-31:
9.10.0 worked well, except for an issue reporting battery state.

It appears that the plugin is calculating percentage based on the minimum SOC. Ie if that is at 10%, then the percentage reported is based on the 90% usable. Battery state 50%, reported percentage 40/90 = 44%. That is fine.

However, the accessory then applies the minimum SOC a second time, and so does not report effective  percentage until the battery reaches 11% effective charge (ie 20% actual charge)

grzegorz914, 2025-03-31:
Try 9.11.0-beta, if backup level is < min.soc then is 0% in other case is scaled 0-100%

Note, after update please configure credentials mode in plugin config.

steve-lad, 2025-03-31:
9.11.0-beta4 fails to load (separate issue reported)

grzegorz914, 2025-04-01:
Any info about latest beta?

steve-lad, 2025-04-01:
9.11.0-beta.8 Works exactly as expected. Battery state is actual state when >MinSoC. No issues.

will switch to beta.11 when have checked all contacts

grzegorz914, 2025-04-02:
If You have some other suggestions/issues let me know in other case I will release 9.11.0 today later.

closed 2025-05-11 by github-actions[bot]