grzegorz914/homebridge-enphase-envoy #193: V9.11.0 Beta 4 fails to Load

steve-lad, 2025-03-31:
```
Error: Device: 192.168.30.150 PV Array, Did finish launching error: ReferenceError: acBatterieBackupLevelSummaryAccessory is not defined.
    at HomebridgeAPI.<anonymous> (file:///usr/local/lib/node_modules/homebridge-enphase-envoy/index.js:148:17)
    at HomebridgeAPI.emit (node:events:518:28)
    at HomebridgeAPI.signalFinished (/usr/local/lib/node_modules/homebridge/src/api.ts:275:10)
    at ChildBridgeFork.startBridge (/usr/local/lib/node_modules/homebridge/src/childBridgeFork.ts:194:14)
[3/31/2025, 10:36:24 AM] [homebridge-enphase-envoy] Child bridge process ended
[3/31/2025, 10:36:24 AM] [homebridge-enphase-envoy] Process Ended. Code: 1, Signal: null
```

grzegorz914, 2025-03-31:
Fixed

steve-lad, 2025-03-31:
9.11.0 Beta 6. Using enlighten credentials. Runs, but no accessories (in homebridge, and not responding in Home)

There is also a typo in the acbatteries section (Individual should read Individual) - just info

grzegorz914, 2025-03-31:
Just add it again to homekit

steve-lad, 2025-03-31:
They do not show in Homebridge, so cannot show in home. I have tried reverting, but cannot find the last working version that I had, and so nothing works now!

grzegorz914, 2025-03-31:
Get the log please

steve-lad, 2025-03-31:
```
[3/31/2025, 2:04:33 PM] [homebridge-enphase-envoy] Restarting Process...
[3/31/2025, 2:04:33 PM] [homebridge-enphase-envoy] Launched child bridge with PID 6965
[3/31/2025, 2:04:33 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.11.0-beta.7 child bridge successfully
[3/31/2025, 2:04:36 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Arf Profile not supported, dont worry all working correct, only the profile name will not be displayed.
[3/31/2025, 2:04:38 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Error: Start error: Error: Update inventory error: AxiosError: Request failed with status code 401, trying again.
```

grzegorz914, 2025-03-31:
Will fix asap

grzegorz914, 2025-03-31:
Fixed

steve-lad, 2025-03-31:
Yes - thanks

closed 2025-03-31 by steve-lad