grzegorz914/homebridge-enphase-envoy #195: Error no more Data

SerrII, 2025-04-18:
Hello, 

I installed the Enphase plugin. The plugin briefly received updates from Enphase. But then an error message quickly appeared and no new data was delivered.

Error:

Device: ??? Enphase, Update production state error: AxiosError: Request failed with status code 401, dont worry all working correct, only the production state monitoring sensor and comtrol will not be displayed.

Thank you very much for your help in advance

grzegorz914, 2025-04-18:
Please enable debug mode in plugin config then post the whole log.

grzegorz914, 2025-04-21:
Still the issue exist with latest release?

thevangogh, 2025-04-22:
plus one with this issue.

grzegorz914, 2025-04-22:
Debug please

grzegorz914, 2025-04-22:
Try 9.14.3-beta.0 and let me know

thevangogh, 2025-04-22:
[log-2.txt](https://github.com/user-attachments/files/19845550/log-2.txt)

grzegorz914, 2025-04-22:
try latest beta and let me know

grzegorz914, 2025-04-23:
v9.14.3 released, try and let me know

thevangogh, 2025-04-25:
Resolved for me.

closed 2025-05-19 by grzegorz914