grzegorz914/homebridge-enphase-envoy #196: AxiosError: Request failed with status code 404

thevangogh, 2025-04-22:
hi there,
I encountered the following error log for build  [9.14.2] - (20.04.2025)
"Error: Start error: Error: Update ensemble inventory error: AxiosError: Request failed with status code 404, trying again."

Seems to be related to "Requesting ensemble inventory."

Any help would be appreciated. thanks

FYI, I had run successfully with the previous build.

grzegorz914, 2025-04-22:
Debug log please.

grzegorz914, 2025-04-22:
Try 9.14.3-beta.0 and let me know

thevangogh, 2025-04-22:
Seems to have resolved the "Requesting ensemble inventory" issue but now have live data update issue.

"06:10:19 PMEnphase Envoy BridgeenphaseEnvoyerrorDevice: 192.168.2.85 Power Production, Error: Start error: Error: Update live data error: AxiosError: Request failed with status code 404, trying again.".

thevangogh, 2025-04-23:
I am using Hoobs unfortunately and shows a list that is not reflected of the latest model. 
All good for now. I've reverted to version 9.12.6 for now. thanks for your helpl

grzegorz914, 2025-04-23:
v9.14.3 released, try and let me know

thevangogh, 2025-04-25:
Resolved. thank you very much. amazing effort.

closed 2025-04-25 by thevangogh