grzegorz914/homebridge-enphase-envoy #198: issues enabling power monitoring

robmbrooks, 2025-05-18:
Looking at the logs my microinverters are seen and power figures come from them, my attempts at getting an accessory reporting the power level have failed.

`    {
      "devices": [
        {
          "name": "Envoy",
          "host": "envoy.localdomain",
          "displayType": 3,
          "envoyFirmware7xxTokenGenerationMode": 1,
          "enlightenUser": "removed-email",
          "enlightenPasswd": "removed-password",
          "envoyTokenInstaller": false,
          "productionStateControl": {
            "displayType": 0,
            "namePrefix": false
          },
          "productionStateSensor": {
            "displayType": 0,
            "namePrefix": false
          },
          "plcLevelControl": {
            "displayType": 0,
            "namePrefix": false
          },
          "powerProductionSummary": 10000,
          "powerProductionStateSensor": {
            "displayType": 2,
            "name": "PowerSummary",
            "namePrefix": true
          },
          "powerProductionLevelSensors": [
            {
              "displayType": 2,
              "name": "PowerLevel",
              "compareMode": 0,
              "powerLevel": 10000,
              "namePrefix": true
            }
          ],`

Logs are attached...

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/20273911/homebridge-enphase-envoy.log.txt)

grzegorz914, 2025-05-19:
Please enable debug mode in plugin config then send me the log

grzegorz914, 2025-05-19:
Please also try 9.15.0-beta and let me know

robmbrooks, 2025-05-19:
I've upgrade to v9.15.0-beta.20 and enabled debug, logs are below...

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/20301946/homebridge-enphase-envoy.log.txt)

grzegorz914, 2025-05-19:
try beta.21 and let me know

robmbrooks, 2025-05-19:
This version looks good! I have working sensors and the logs look happier. Let me know if you'd like to me to share the current logs.

grzegorz914, 2025-05-19:
yes, post the log please

robmbrooks, 2025-05-19:
new logs:

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/20303098/homebridge-enphase-envoy.log.txt)

robmbrooks, 2025-05-19:
I installed v9.15.1, but the logs show this...

[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production power: NaN kW.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production power peak: NaN kW.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production power peak detected: No.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production energy today: NaN kWh.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production energy last seven days: NaN kWh.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production energy lifetime: NaN kWh.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production last report: undefined.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production power peak reset: Off.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production power state sensor: Not active.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] Device: envoy.localdomain Envoy, Production power level sensor: Envoy PowerLevel: Not active.
[5/19/2025, 8:39:17 PM] [homebridge-enphase-envoy] This plugin generated a warning from the characteristic 'Current Relative Humidity': characteristic value expected valid finite number and received "NaN" (number). See https://homebridge.io/w/JtMGR for more info.

reverting to the beta cleared the issue...

grzegorz914, 2025-05-19:
Fixed in 9.15.2

robmbrooks, 2025-05-19:
Confirmed! Thank you!

closed 2025-05-19 by grzegorz914