grzegorz914/homebridge-enphase-envoy #199: Consumption net values are not working

Ricadri, 2025-05-26:
Hello,

Since my last update (9.16) I have an issue with my Consumption net power sensors :

"powerConsumptionNetLevelSensors": [
                {
                    "displayType": 1,
                    "name": "Conso net 200W",
                    "compareMode": 1,
                    "powerLevel": 200,
                    "namePrefix": false
                },
                {
                    "displayType": 1,
                    "name": "Conso net 2kW",
                    "compareMode": 1,
                    "powerLevel": 2000,
                    "namePrefix": false
                },
                **{
                    "displayType": 1,
                    "name": "Injection 1kW",
                    "compareMode": 4,
                    "powerLevel": -1000,
                    "namePrefix": false
                },
                {
                    "displayType": 1,
                    "name": "Injection 2kW",
                    "compareMode": 4,
                    "powerLevel": -2000,
                    "namePrefix": false
                },**
                {
                    "displayType": 1,
                    "name": "Injection 3kW",
                    "compareMode": 4,
                    "powerLevel": -3000,
                    "namePrefix": false
                },
                {
                    "displayType": 1,
                    "name": "Injection 200W",
                    "compareMode": 4,
                    "powerLevel": -200,
                    "namePrefix": false
                }
            ],

The sensors are not anymore active even when I exporting a lot of power.

Here is logs examples :

[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net power: **-2.273815 kW.**
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net power peak: 0 kW.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net power peak detected: No.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net energy today: 0 kWh.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net energy last seven days: 0 kWh.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net energy lifetime: 1564.23375 kWh.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net reactive power: 0.735171 kVAr.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net apparent power: -2.5478389999999997 kVA.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net current: -10.561 A.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net voltage: 241.253 V.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net frequency: 50.125 Hz.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net power factor: -1 cos φ.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net last report: 5/26/2025, 3:47:26 PM.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption Net power peak reset: Off.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption net power level sensor: Conso net 200W: Not active.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption net power level sensor: Conso net 2kW: Not active.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption net power level sensor: **Injection 1kW: Not active.**
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption net power level sensor: **Injection 2kW: Not active.**
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption net power level sensor: Injection 3kW: Not active.
[5/26/2025, 3:47:16 PM] [homebridge-enphase-envoy] Device: 192.168.1.130 Panneaux, Consumption net power level sensor: **Injection 200W: Not active.**

with this example "Injection 1kW" and "Injection 2kW" should be active.

Can you help me ?

Ricadri, 2025-05-26:
[homebridge-enphase-envoy.log (1).txt](https://github.com/user-attachments/files/20442803/homebridge-enphase-envoy.log.1.txt)

Here is the debug logs

grzegorz914, 2025-05-26:
Thanks for report, fixed in v9.17.1

closed 2025-05-26 by grzegorz914