grzegorz914/homebridge-enphase-envoy #201: no accessories shown despite successful bridge addition

nebosphere19, 2025-06-06:
I've added the bridge but the lightbulb accessory that is supposed to show production of my panels never appears in Home app. I've even selected the room. It all appears to be working in the logs, no errors, etc, etc. I've attempted to reset the cached accessories for this plugin in Homebridge, I've verified the IP address of my gateway. Not sure what else to do. 

> [6/6/2025, 9:54:38 AM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.19.1 child bridge successfully
[6/6/2025, 9:54:38 AM] Loaded 0 cached accessories from cachedAccessories.0EE7E7DFA6D5.
[6/6/2025, 9:54:38 AM] Homebridge v1.9.0 (HAP v0.12.3) (homebridge-enphase-envoy) is running on port 47387.
[6/6/2025, 9:54:38 AM] 

NOTICE TO USERS AND PLUGIN DEVELOPERS
> Homebridge 2.0 is on the way and brings some breaking changes to existing plugins.
> Please visit the following link to learn more about the changes and how to prepare:
> https://github.com/homebridge/homebridge/wiki/Updating-To-Homebridge-v2.0

[6/6/2025, 9:54:38 AM] [homebridge-enphase-envoy] Device: 10.0.0.64 Solar Panels, Cookie not valid, refreshing.
[6/6/2025, 9:54:39 AM] [homebridge-enphase-envoy] Device: 10.0.0.64 Solar Panels, Cookie refresh success.
[6/6/2025, 9:54:44 AM] [homebridge-enphase-envoy] Device: 10.0.0.64 Solar Panels, Connect Success.
[6/6/2025, 9:54:44 AM] [homebridge-enphase-envoy] Device: 10.0.0.64 Solar Panels, Published as external accessory.
[6/6/2025, 9:54:44 AM] [homebridge-enphase-envoy] Device: 10.0.0.64 Solar Panels, Impulse generator started.
[6/6/2025, 9:54:44 AM] Solar Panels 9178 is running on port 62629.
[6/6/2025, 9:54:44 AM] Please add [Solar Panels 9178] manually in Home app. Setup Code: 852-81-469

[image]

[image]

grzegorz914, 2025-06-06:
https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#adding-to-homekit

grzegorz914, 2025-06-06:
You don’t need add the bridge, only the accessory itself. Try this https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting

nebosphere19, 2025-06-06:
I was trying to add the accessory, but it just added the bridge (the first time). So I've tried what you mentioned above and now the accessory no longer shows up in homebridge either (it did before). It added the bridge again, but no accessory, just like last time.

nebosphere19, 2025-06-06:
> You need to add the accessory with name Solar Panels not the bridge.

I would do that, except it's not an option. There's no accessory listed
from the Envoy/Enphase system.

grzegorz914, 2025-06-07:
Can you get a screenshot or video ho do You do that?

nebosphere19, 2025-06-07:
Sure! see attached screenshots for the step by step.

[image]
[image]
[image]
[image]
[image]
[image]

grzegorz914, 2025-06-07:
Ok, first remove the bridge from Home app first, next remove the bridge and accessory from Homebridge with this: https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting NEXT restart whole Homebridge and try to add the accessory again, do not add the enphase bridge.

nebosphere19, 2025-06-08:
Did that. Unfortunately, the only item I'm presented with is the bridge.
there is no accessory to add. see attached screenshot.

nebosphere19, 2025-06-20:
I think I'm probably just going to have to accept that this doesn't work for me, which is unfortunate! LMK if you have any other suggestions....

grzegorz914, 2025-06-20:
Just remove all enphase accesory and bridges from Home.app then restart Your phone, next remove the bridge and accessory from Homebridge with this: https://github.com/grzegorz914/homebridge-enphase-envoy/wiki#troubleshooting next restart whole Homebridge and try to add the accessory again, newer add the enphase bridge. If you can please make a video how did you that.

nebosphere19, 2025-06-20:
I've tried all of this. I even removed the entire plugin and started again from scratch. As I showed before, it does not present an accessory for me to add to the Home app. It only presents the bridge itself. Is there something I'm missing in the plugin config?

nebosphere19, 2025-07-07:
for some reason it's working now after upgrading the plugin. The only strange thing is the plugin still shows that it's not connected to HomeKit, even though it clearly is since I can see the lightbulb for production there and it's updating as the sun goes in and out of the clouds.

closed 2025-08-15 by github-actions[bot]