grzegorz914/homebridge-enphase-envoy #203: Update issue

bbleyzat, 2025-06-16:
Hello,

Since yesterday, I’ve been experiencing issues with the plug-in, the data is not refreshing correctly.
I’ve tried several steps, including rolling back to previous versions and updating again, but the issue persists.

If I reboot the plug-in I got the right value, but after it is stuck.

[Homebridge log.txt](https://github.com/user-attachments/files/20753697/Homebridge.log.txt)

Thanks in advance for your help.

bbleyzat, 2025-06-16:
> which data is not updated

The value in % of production

bbleyzat, 2025-06-16:
Thanks for the tips. I am cleaning, I'll keep you posted

bbleyzat, 2025-06-16:
So I did what you told me, it seens to act the same.
I remove the accessories, clean old installation still paired. Remove the child bridge, re-install and add to homekit again.
The % of production is still suck.

grzegorz914, 2025-06-16:
What is your production refresh time

bbleyzat, 2025-06-16:
[image]
We discussed once about this settings. Are they still valid?

bbleyzat, 2025-06-16:
And this % is stuck, and I don't see any kind of error in the log. Working well during the past months. So weird.

bbleyzat, 2025-06-16:
Hey, unfortunately same issue. Still suck, I rebooted the plug in and after it show the correct value, and stuck again.

bbleyzat, 2025-06-16:
`[6/16/2025, 4:55:31 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production level: 53 %.
[6/16/2025, 4:55:45 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 4:56:04 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:00:09 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:00:09 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production level: 53 %.
[6/16/2025, 5:00:19 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:00:21 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:08:44 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:08:45 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:08:46 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:08:46 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:09:17 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:09:17 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:10:13 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:10:14 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:10:44 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:10:45 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:11:26 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:11:26 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:11:56 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:11:57 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:12:50 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:12:51 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:13:03 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:13:04 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:13:33 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:13:33 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:13:34 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:13:34 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production level: 53 %.
[6/16/2025, 5:13:41 PM] [homebridge-enphase-envoy] Restarting child bridge...
[6/16/2025, 5:13:41 PM] Got SIGTERM, shutting down child bridge process...
[6/16/2025, 5:13:46 PM] [homebridge-enphase-envoy] Child bridge process ended
[6/16/2025, 5:13:46 PM] [homebridge-enphase-envoy] Process Ended. Code: 143, Signal: null
[6/16/2025, 5:13:53 PM] [homebridge-enphase-envoy] Restarting Process...
[6/16/2025, 5:13:53 PM] [homebridge-enphase-envoy] Launched child bridge with PID 19346
[6/16/2025, 5:13:54 PM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[6/16/2025, 5:13:54 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v9.20.2-beta.18 child bridge successfully
[6/16/2025, 5:13:54 PM] Loaded 0 cached accessories from cachedAccessories.0E6CC09D69D8.
[6/16/2025, 5:13:54 PM] Homebridge v1.10.0 (HAP v0.13.1) (Enphase) is running on port 38540.
[6/16/2025, 5:13:54 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Token exist but not valid, validating.
[6/16/2025, 5:13:55 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Token validate success.
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Connect Success.
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] -------- Énergie solaire --------
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Manufacturer: 
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Model: ENV-S-WM-230
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Firmware: D8.2.4382
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] SerialNr: 122324003500
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Time: 6/16/2025, 5:13:53 PM
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] ------------------------------
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Q-Relays: 1
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Inverters: 10
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Wireless Kit: 1
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] --------------------------------
[6/16/2025, 5:14:03 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Published as external accessory.
[... cut ...]
[6/16/2025, 5:14:11 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:16:31 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:16:31 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production level: 48 %.
[6/16/2025, 5:16:45 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.
[6/16/2025, 5:16:46 PM] [homebridge-enphase-envoy] Device: envoy.local Énergie solaire, Production state: Enabled.`

bbleyzat, 2025-06-16:
[extract_log_before_after_reboot.pdf](https://github.com/user-attachments/files/20760003/extract_log_before_after_reboot.pdf)

grzegorz914, 2025-06-16:
Get me the Debug again

bbleyzat, 2025-06-16:
Sure.

[Homebridge log_1_debug.txt](https://github.com/user-attachments/files/20760088/Homebridge.log_1_debug.txt)

grzegorz914, 2025-06-16:
Try again latest beta

bbleyzat, 2025-06-16:
Same behavior.

grzegorz914, 2025-06-16:
Please again the debug

bbleyzat, 2025-06-16:
Yes, sorry!

[Homebridge log_debug_2.txt](https://github.com/user-attachments/files/20761569/Homebridge.log_debug_2.txt)

bbleyzat, 2025-06-16:
I tried b22 if I need to rollback don't hesitate to tell me.

[Homebridge log.txt](https://github.com/user-attachments/files/20762693/Homebridge.log.txt)

bbleyzat, 2025-06-16:
Sure, I kept device info log + info log

[Homebridge log.txt](https://github.com/user-attachments/files/20763044/Homebridge.log.txt)

bbleyzat, 2025-06-16:
Oh yeah!! From 1% to 0!! Seems to work! I’ll confirm again tomorrow morning

bbleyzat, 2025-06-17:
[Homebridge 7B28 log_DEBUG_LAST_DANCE.txt](https://github.com/user-attachments/files/20769147/Homebridge.7B28.log_DEBUG_LAST_DANCE.txt)

Here they are.

closed 2025-06-19 by bbleyzat