grzegorz914/homebridge-enphase-envoy #206: Start error: Error: Prepare accessory error: TypeError: Cannot read properties of undefined (reading 'onGet'), trying again.

vypr-zz, 2025-07-03:
Hi, I'm getting an error during plugin startup and Im not sure which accessory type it's erroring on.

`Device: envoy.local Enphase Envoy, Error: Start error: Error: Prepare accessory error: TypeError: Cannot read properties of undefined (reading 'onGet'), trying again.`

Im running v10.1.0 of the plugin on Homebridge v1.10.0, Homebridge UI v4.77.0, and Node JS v22.17.0.

I've attached a copy of my log file in debug mode

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21029539/homebridge-enphase-envoy.log.txt)

vypr-zz, 2025-07-04:
On beta 40, I get this error `Error: Start error: Error: Update tariff error: TypeError: Cannot read properties of undefined (reading 'mode'), trying again.`
Here's the attached debug log

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21060776/homebridge-enphase-envoy.log.txt)

grzegorz914, 2025-07-04:
Try new beta please

vypr-zz, 2025-07-04:
Beta 46 logs

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21064425/homebridge-enphase-envoy.log.txt)

grzegorz914, 2025-07-04:
Many thanks, beta 48 fixed also characteristics warning

vypr-zz, 2025-07-04:
Beta 50 logs

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21068684/homebridge-enphase-envoy.log.txt)

grzegorz914, 2025-07-05:
OK, b7 should fixed all rest problems

vypr-zz, 2025-07-05:
beta 8 debug logs, with error `Start error: Error: Update ensemble data error: Cannot read properties of undefined (reading 'updateCharacteristic'), trying again.`

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21081747/homebridge-enphase-envoy.log.txt)

vypr-zz, 2025-07-06:
Debug logs from beta 10

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21089316/homebridge-enphase-envoy.log.txt)

grzegorz914, 2025-07-06:
Thanks, all good, some small changes in latest beta to test

hkaushal, 2025-08-17:
After updating to latest Beta. The issue got resolved but a new issue appeared. `Device: 192.168.5.33 Test, Error: Start error: Error: Update home data error: enchg is not defined, retrying.` 

Thanks for keeping this plugin.

hkaushal, 2025-08-17:
Thank you so much. Now its throwing below error
`Device: 192.168.5.33 Test, Error: Start error: Error: Update home data error: enchg is not defined, retrying.`

closed 2025-10-03 by grzegorz914