grzegorz914/homebridge-enphase-envoy #207: Encharge battery backup level incorrect (intermittently) v10.1.0

steve-lad, 2025-07-08:
Hi,

I have just upgraded to 10.1.0, and am seeing incorrect encharge battery backup level.  This also shows in the logs, so I am not sure where the issue is. This was never the case with prior versions. 

This is intermittent - it is as if it is reading or calculating erratically!

grzegorz914, 2025-07-09:
try latest beta and let me know

grzegorz914, 2025-07-09:
fixed with 178

steve-lad, 2025-07-16:
Beta 284

Various contact sensors and calculated values are wrong, including battery

eg

[7/16/2025, 4:12:45 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Production level: 35 %
[7/16/2025, 4:12:49 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, power: 3.821223 kW

Power is near maximum , but reported at 35%

```
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, power: 4.050231 kW
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, power state: On
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, power level: 100 %
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, power peak detected: No
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, energy today: 27.012917 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, energy last seven days: 3087.811917 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, energy lifetime: 3110.808008 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, energy state: Enabled
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, current: 17.586 A
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, voltage: 246.04933333333335 V
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, power factor: 0.94 cos φ
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Production, frequency: 50 Hz
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, power: -1.3700640000000002 kW
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, power state: Off
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, power level: 0 %
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, power peak detected: No
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, energy today: 0 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, energy last seven days: 0 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, energy lifetime: 700.2727389999999 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, energy state: Disabled
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, current: -5.136 A
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, voltage: 246.15533333333335 V
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, power factor: -1 cos φ
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Net, frequency: 50 Hz
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, power: 2.680167 kW
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, power state: Off
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, power level: 0 %
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, power peak detected: No
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, energy today: 24.504 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, energy last seven days: 3831.869 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, energy lifetime: 3814.3780699999998 kWh
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, energy state: Enabled
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, current: 12.45 A
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, voltage: 246.15533333333335 V
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, power factor: 0.88 cos φ
[7/16/2025, 3:43:26 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power And Energy: Consumption Total, frequency: 50 Hz
[... cut ...]
[7/16/2025, 3:43:28 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, IQBattery backup level: 13 %
[7/16/2025, 3:43:28 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, IQBattery state: Charged
[7/16/2025, 3:43:28 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, IQBattery Backup Level sensor: Backup  > 85%, state: Not active
[7/16/2025, 3:43:28 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Power Level: Production, sensor: Production Level > 3Kw: Not active
```

same issue, but now the contact sensor is also wrong

grzegorz914, 2025-07-25:
Try latest beta and let me know

grzegorz914, 2025-07-26:
Try again, look only on PV production level

steve-lad, 2025-07-26:
I did, it was not stable.  I have automations that work based on these levels, and this instability causes issues.

Prior to 10.x, both of the levels worked perfectly.

steve-lad, 2025-07-27:
Beta 319

` Device: 192.168.30.150 PV Array, Error: Start error: Error: Update power and energy data error: Cannot access 'obj' before initialization, retrying.`

grzegorz914, 2025-07-27:
Fixed

steve-lad, 2025-07-27:
Beta 320

For the moment, PV is stable. Could the issue be that the instability showed when PV was 100%? - there was definitely an oscillation, although it may not have been captured in the logs.

Separate issue, unrelated: Both charge profiles are on and cannot be closed in this beta. This should be  captured in the attached log.

[homebridge.log.txt](https://github.com/user-attachments/files/21452761/homebridge.log.txt)

steve-lad, 2025-07-27:
Beta 320

at 11:25, PV drops from around 80% to 30% and then back to around 80%

[homebridge.log.txt](https://github.com/user-attachments/files/21453081/homebridge.log.txt)

grzegorz914, 2025-07-27:
try latest beta and check only charge profile issue reported in previous post

grzegorz914, 2025-07-27:
Try 322 the issue should be fixed

grzegorz914, 2025-07-27:
Try again 324

steve-lad, 2025-07-27:
Beta 324 

17:46 Change from ~85% to ~36% for 2 readings (other cases seen but this one noted)

[homebridge.log.txt](https://github.com/user-attachments/files/21454836/homebridge.log.txt)

grzegorz914, 2025-07-27:
All is correct, power at this time was 1436W ~36%, plugin working correctly.

grzegorz914, 2025-07-28:
Test latest beta and get the log please

grzegorz914, 2025-08-28:
Still the issue exist with latest beat?

grzegorz914, 2025-09-26:
any info?

aapatell, 2025-10-01:
Hello! thanks of much for this plugin, I recently got solar installed and this is the first thing I did. I have a 2x IQ Battery 10C system.

I believe I am having a similar issue so adding here rather than a new topic. All my accessories and sensors are working except the battery backup level. It only updates when I restart homebridge and then stays at the same percentage until I restart again. 

[image]

[image]

[image]

[image]

[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/22645087/homebridge-enphase-envoy.log.txt)

grzegorz914, 2025-10-01:
Please try 10.2 beta and get the log please

grzegorz914, 2025-10-01:
Just open plugin settings UI press save button and restart the plugin

aapatell, 2025-10-01:
ah, thats simple. thank you

Version 452 and 453 were not working for me. kept getting an error message in the logs. 

[homebridge-enphase-envoy.log-3.txt](https://github.com/user-attachments/files/22646087/homebridge-enphase-envoy.log-3.txt)

grzegorz914, 2025-10-01:
453 need to be working

aapatell, 2025-10-01:
Ok, I’ll pop back to 453 tonight and circle back.

grzegorz914, 2025-10-02:
458 warning fixed, please enable debug in plugin config advanced>>settings>log an then get the log please

aapatell, 2025-10-02:
beta 458, production error returned. 

updated log with debug:

[homebridge-enphase-envoy.log-5.txt](https://github.com/user-attachments/files/22663083/homebridge-enphase-envoy.log-5.txt)

.

aapatell, 2025-10-02:
beta 467, running well

logs:

[homebridge-enphase-envoy.log-7.txt](https://github.com/user-attachments/files/22663737/homebridge-enphase-envoy.log-7.txt)

closed 2025-10-03 by grzegorz914