grzegorz914/homebridge-enphase-envoy #208: Enhancement Request: Add ability to activate/deactivate charge from grid for encharges

steve-lad, 2025-07-08:
v10.1.0

I have not found a setting (switch) that will allow me to activate/deactivate the charge from grid function for encharge battery. I see that there is a section that mentions control, but they are all only sensors. Apologies if I have missed something......

grzegorz914, 2025-07-08:
Was bug in UI, try latest beta and go to:

[image]

grzegorz914, 2025-07-08:
In latest beta I have added possibility to set encharge profile and allow charge from grid:

[image]

steve-lad, 2025-07-08:
Thanks.

I think that there is another obstacle.

Selecting Full BackUp in the first beta, briefly showed a light bulb, which disappeared. The probable reason is that full backup is not allowed in France, and so the grid profile overwrote.

With the second beta, nothing appeared. A possible explanation is that via the web UI, there is a pop up that appears to require you to confirm that this is allowed, and then a period selector. Screenshots from the web UI attached as you may have a different system.

[image]

What I would like is to have profile self consumption, with charge from grid as an option.

grzegorz914, 2025-07-08:
I have integrate it into the profile, if You don't see the lightbulbb tile please send me the debug log, enable it in Advanced section>>>System>>>Log debug

steve-lad, 2025-07-08:
Still no light bulb. It shows in homebridge, but immediately offs on refresh, and has no impact in enlighten manager.

I have replied directly with the relevant debug log

steve-lad, 2025-07-08:
> grzegorz914
>  left a comment
> (grzegorz914/homebridge-enphase-envoy#208)
> tru beta 168 the lightbulb should be available again

[7/8/2025, 9:02:20 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Tariff: {
  "tariff": {
    "currency": {
      "code": "EUR"
    },
    "logger": "mylogger",
    "date": "1741693935",
    "storage_settings": {
      "mode": "self-consumption",
      "operation_mode_sub_type": "",
      "reserved_soc": 0,
      "very_low_soc": 5,
      "charge_from_grid": false,
      "date": "1741693935",
      "opt_schedules": true
    },
    "single_rate": {
      "rate": 0.20160000026226044,
      "sell": 0
    },
    "seasons": [
      {
        "id": "all_year_long",
        "start": "1/1",
        "days": [
          {
            "id": "all_days",
            "days": "Mon,Tue,Wed,Thu,Fri,Sat,Sun",
            "must_charge_start": 0,
            "must_charge_duration": 0,
            "must_charge_mode": "CP",
            "peak_rule": "DL",
            "enable_discharge_to_grid": false,
            "periods": [
              {
                "id": "period_1",
                "start": 0,
                "rate": 0.20160000026226044
              }
            ]
          }
        ],
        "tiers": []
      }
    ],
    "seasons_sell": []
  },
  "schedule": {
    "filename": "",
    "source": "",
    "date": "2025-07-08 19:06:28 UTC",
    "version": "",
    "reserved_soc": 30,
    "very_low_soc": 10,
    "charge_from_grid": false,
    "batt_mode": "self-consumption",
    "battery_mode": "self-consumption",
    "operation_mode_sub_type": "",
    "opt_schedules": false,
    "override": false,
    "override_backup_soc": 30,
    "override_chg_dischg_rate": 0,
    "override_tou_mode": "StorageTouMode_DEFAULT_TOU_MODE",
    "schedule": {
      "Disable": [
        {
          "Sun": [
            {
              "start": 0,
              "duration": 1440,
              "setting": "ID"
            }
          ]
        },
        {
          "Mon": [
            {
              "start": 0,
              "duration": 1440,
              "setting": "ID"
            }
          ]
        },
        {
          "Tue": [
            {
              "start": 0,
              "duration": 1440,
              "setting": "ID"
            }
          ]
        },
        {
          "Wed": [
            {
              "start": 0,
              "duration": 1440,
              "setting": "ID"
            }
          ]
        },
        {
          "Thu": [
            {
              "start": 0,
              "duration": 1440,
              "setting": "ID"
            }
          ]
        },
        {
          "Fri": [
            {
              "start": 0,
              "duration": 1440,
              "setting": "ID"
            }
          ]
        },
        {
          "Sat": [
            {
              "start": 0,
              "duration": 1440,
              "setting": "ID"
            }
          ]
        }
      ]
    }
  }
}
[7/8/2025, 9:02:20 PM] [homebridge-enphase-

steve-lad, 2025-07-08:
It looks like the charge periods are absent?

grzegorz914, 2025-07-08:
I need the whole debug, download the debug file and post here

steve-lad, 2025-07-08:
￼

> 
> 
> grzegorz914
>  left a comment 
> (grzegorz914/homebridge-enphase-envoy#208)
> I need the whole debug, download the debug file and post here

[7/8/2025, 9:01:16 PM] [homebridge-enphase-envoy] Restarting Process...
[7/8/2025, 9:01:16 PM] [homebridge-enphase-envoy] Launched child bridge with PID 87679
[7/8/2025, 9:01:17 PM] Registering platform 'homebridge-enphase-envoy.enphaseEnvoy'
[7/8/2025, 9:01:17 PM] [homebridge-enphase-envoy] Loaded homebridge-enphase-envoy v10.2.0-beta.167 child bridge successfully
[7/8/2025, 9:01:17 PM] Loaded 0 cached accessories from cachedAccessories.0E4A5C651231.
[7/8/2025, 9:01:17 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, did finish launching.
[7/8/2025, 9:01:17 PM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, Config: {
  "name": "PV Array",
  "host": "192.168.30.150",
  "displayType": 1,
  "envoyFirmware7xxTokenGenerationMode": 1,
  "envoyPasswd": "removed",
  "enlightenUser": ***@***.***",
  "enlightenPasswd": "removed",
  "envoyTokenInstaller": false,
  "lockControl": true,
  "lockControlPrefix": false,
  "energyMeter": true,
  "productionStateSensor": {
    "displayType": 0,
    "namePrefix": false
  },
  "plcLevelControl": {
    "displayType": 0,
    "namePrefix": false
  },
  "powerProductionSummary": 4050,
  "powerProductionLevelSensors": [
    {
      "displayType": 0,
      "compareMode": 0,
      "namePrefix": false
    },
    {
      "displayType": 3,
      "name": "Production Level > 3Kw",
      "compareMode": 0,
      "powerLevel": 3000,
      "namePrefix": false
    }
  ],
  "energyProductionLevelSensors": [
    {
      "displayType": 0,
      "compareMode": 0,
      "energyLevel": 0,
      "namePrefix": false
    }
  ],
  "energyProductionLifetimeOffset": 0,
  "gridProductionQualitySensors": [
    {
      "displayType": 0,
      "compareType": 0,
      "compareMode": 0,
      "compareLevel": 0,
      "namePrefix": false
    }
  ],
  "powerConsumptionTotalLevelSensors": [
    {
[... 2305 lines cut ...]
          "acCurrentInmA": 29,
          "l1NAcVoltageInmV": 0,
          "l2NAcVoltageInmV": 0,
          "l3NAcVoltageInmV": 0,
          "rssi": 106,
          "issi": 56
        },
        "lifetime": {
          "createdTime": 1751517986,
          "duration": 16795792,
          "joulesProduced": 1124137350
        }
      }
    ]
  },
  "553649408": {
    "devName": "pcu",
    "sn": "122417014174",
    "active": true,
    "modGone": false,
    "channels": [
      {
        "chanEid": 1627391249,
        "created": 1752001432,
        "wattHou

grzegorz914, 2025-07-09:
The encharge profile control tile is published, it must be there in home app:

[image]

steve-lad, 2025-07-09:
After deleting and reinstalling the bridge, the bulb is now there. At the moment, it is impossible to verify that it works, as there is PV production and I believe that the system will disable charge from grid when there is production (the bulb goes on then immediately off). I will try this evening and let you know

steve-lad, 2025-07-10:
Beta 194

Charge from grid is not working as intended. Are there more settings that need to be enabled? To help, here are 2 log files in debug mode.

homebridge.log-2.txt
Charge from grid via homebridge. Light goes on and then off. No charging

homebridge.log-3.txt
Charge from grid via enlighten. Charging starts after a short delay

[homebridge.log-2.txt](https://github.com/user-attachments/files/21155331/homebridge.log-2.txt)

[homebridge.log-3.txt](https://github.com/user-attachments/files/21155337/homebridge.log-3.txt)

grzegorz914, 2025-07-10:
i dont se that point You set the profile, try again with latest beta and home app and get me the log

steve-lad, 2025-07-10:
Beta 206

Accessory name changed so profile should be visible

[homebridge.log.txt](https://github.com/user-attachments/files/21160029/homebridge.log.txt)

grzegorz914, 2025-07-10:
try 207, 
please direct after update plugin cut the log before download from console

steve-lad, 2025-07-10:
Beta 207

After unlocking System Control, Charge battery now stays on. However, it is now not possible to turn it off! It even remains on after a restart.

[homebridge.log-2.txt](https://github.com/user-attachments/files/21162437/homebridge.log-2.txt)

I do not see any changes in Enlighten, nor is the battery charging. This could be because:

High PV production
The hours for charging battery programmed in enlighten

steve-lad, 2025-07-10:
Beta 210

` Device: 192.168.30.150 PV Array, Error: Start error: Error: Prepare accessory error: ReferenceError: ensempcuble is not defined, retrying.`

will try 209

grzegorz914, 2025-07-10:
OK, send me the log after You try to change the battery profile

grzegorz914, 2025-07-10:
You can create for example 2 profile 
1. Self-Consumptions with Charge from Grid ON
2. Self-Consumption with Charge from Grid OF
 so, the switch is activated based on current state, if encharge have profile Self-Consumption and Charge From Grid is ON then 1 tile is ON and all other are OFF.

The lightbulb can set reserve SOC 0-100%, also the lightbulb shows actual settings of reserve SOC.

grzegorz914, 2025-07-10:
Yes, every operation need unlock system, lock will automatically after 30sec. You can disable this function in plugin settings.

grzegorz914, 2025-07-11:
Hello, if You can please test latest beta and post the log please

steve-lad, 2025-07-11:
OK will do. Beta 212 does not work. The bulbs function correctly, bu even an empty battery with negligible PV does not charge from grid. The enlighten app does not show charge from grid enabled.

Could this be due to the times defined in the enlighten app (the trial was outside of these times?

will try the latest beta tomorrow morning

grzegorz914, 2025-07-12:
Hi, I see in Your config AC Battery enabled
` "acBatterieName": "IQBattery",
  "acBatterieBackupLevelSummaryAccessory": {
    "displayType": 1,
    "minSoc": 10,
    "namePrefix": false
  }, `
but You don’t have it in Your system (set display type to 0), You have encharges (IQ Battery). 
AC Battery was in old system with Envoy FW <= 5.x.x.

steve-lad, 2025-07-12:
I have removed the AC Battery .

Beta 238.

Attached log covers multiple events. Grid Charge on, grid Charge Off enlighten grid charge on, enlighten grid charge off, Home grid charge ON

The log is long, but the state changes show via the System Control log entries

Via Home, no charging. Via Enlighten, charging starts after about 1 min

Hope that this helps

[homebridge.log.txt](https://github.com/user-attachments/files/21193034/homebridge.log.txt)

steve-lad, 2025-07-12:
Some additional information: In enlighten, battery charge period is set to end at 8:40 am. I cannot find this anywhere in the log. Is it possible that there is another information block that you are not reading?

grzegorz914, 2025-07-12:
Fixed

steve-lad, 2025-07-12:
Beta 246

`Device: 192.168.30.150 PV Array, Error: Start error: Error: Update ensemble data error: index is not defined, retrying.`

steve-lad, 2025-07-12:
Beta 248

`This plugin generated a warning from the characteristic 'Reading time': characteristic value expected string and received undefined. See https://homebridge.io/w/JtMGR for more info.`

Not charging from grid, but have PV production. Time window set to allow in Enphase 6:40Pm finish

[homebridge.log.txt](https://github.com/user-attachments/files/21196388/homebridge.log.txt)

steve-lad, 2025-07-14:
Beta 272 - works but has several errors, specifically

`Device: 192.168.30.150 Energy Meter, Error: Start error: Error: Update production pdm error: TypeError: Cannot read properties of undefined (reading 'timeZone'), retrying.`

but also syntax errors

[homebridge.log.txt](https://github.com/user-attachments/files/21212218/homebridge.log.txt)

grzegorz914, 2025-07-14:
Energy meter fixed

steve-lad, 2025-07-14:
I have been looking though the logs, and comparing data with the enlighten app.

You are enabling charge from grid in the tariff block, but also have opt_schedules:true.

In the schedule block, charge_from_grid is false, and schedules is disabled.

Is it possible to enable charge from grid in the schedule block, and to enable the 24hr schedule, to see if this has an impact?

grzegorz914, 2025-07-14:
Added l, try latest beta

steve-lad, 2025-07-14:
Everything is false, but you can see that the battery is charging. Is there a batteries setting block, as in their API, that you are not requesting, as there is nothing in the data below to indicate that it is charging from grid?

[homebridge.log.txt](https://github.com/user-attachments/files/21216447/homebridge.log.txt)

steve-lad, 2025-07-14:
Here is a sequence of screenshots from enlighten for charging from grid, in case you do not have

[image]

[image]

[image]

steve-lad, 2025-07-14:
Found this which includes somme additional parameters

https://support.enphase.com/s/question/0D5Ps00000VDXvZKAX/tldr-i-want-to-be-able-to-manually-start-charging-of-my-5p-batteries-either-in-the-enlighten-app-or-via-the-api-see-more-details-below-current-profiles-dont-provide-enough-flexibility

steve-lad, 2025-07-15:
Beta 284

Loads and runs correctly until set Charge From Grid

```
[7/15/2025, 9:18:57 AM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: Set encharge settings:
[7/15/2025, 9:18:57 AM] [homebridge-enphase-envoy] Device: 192.168.30.150 PV Array, debug: IQBattery set profile: Grid Charge, charge from grid: ON
AxiosError: Request failed with status code 400
    at settle (file:///usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/settle.js:19:12)
    at IncomingMessage.handleStreamEnd (file:///usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/adapters/http.js:599:11)
    at IncomingMessage.emit (node:events:530:35)
    at endReadableNT (node:internal/streams/readable:1698:12)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)
    at Axios.request (file:///usr/local/lib/node_modules/homebridge-enphase-envoy/node_modules/axios/lib/core/Axios.js:45:41)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
```

[homebridge.log.txt](https://github.com/user-attachments/files/21227789/homebridge.log.txt)

grzegorz914, 2025-07-29:
Try latest beta to set the profile

steve-lad, 2025-07-30:
Beta 341

` Device: 192.168.30.150 PV Array, Error: Start error: Error: Update ensemble inventory error: Cannot set properties of undefined (setting 'installed'), retrying.`

grzegorz914, 2025-07-30:
Fixed

grzegorz914, 2025-08-09:
Fixed

grzegorz914, 2025-08-13:
fixed

grzegorz914, 2025-08-16:
Please try latest beta and post the log

steve-lad, 2025-08-17:
Beta 414

Does not charge from grid, and both PV and Grid charge show on with varying percentages

[homebridge.log.txt](https://github.com/user-attachments/files/21820721/homebridge.log.txt)

closed 2025-11-05 by github-actions[bot]