grzegorz914/homebridge-enphase-envoy #209: MQTT messages not being sent

loadedtech, 2025-07-20:
Firstly thanks for your work creating the plugin ... it is appreciated!

I have upgraded to v10.1.0 running on Homebridge on an rPi and can't find any MQTT messages on my broker after trying many config checks/adjustments and much searching. 

The plugin thinks it's connected and is getting data from the Envoy but no messages are hitting the MQTT broker.  Some excerpts from the logs are below.

The broker is Mosquito, which is also on a rPi and working fine for other clients:
    mosquitto version 1.3.4 (build date 2018-09-28 22:21:32+0000)
    mosquitto is an MQTT v3.1 broker.

Device: 172.20.x.y Enphase, debug: Live data: {
  "connection": {
    "mqtt_state": "connected",
    "prov_state": "configured",
    "auth_state": "ok",
    "sc_stream": "enabled",
    "sc_debug": "disabled"
  },
  "meters": {
    "last_update": 1752971564,
    "soc": 0,
    "main_relay_state": 1,
    "gen_relay_state": 5,
    "backup_bat_mode": 1,
    "backup_soc": 30,
    "is_split_phase": 0,
    "phase_count": 3,
    "enc_agg_soc": 0,
    "enc_agg_energy": 0,
    "acb_agg_soc": 0,
    "acb_agg_energy": 0,
    "pv": {
      "agg_p_mw": 7615496,
      "agg_s_mva": 7620975,
      "agg_p_ph_a_mw": 2736433,
      "agg_p_ph_b_mw": 2492867,
      "agg_p_ph_c_mw": 2386196,
      "agg_s_ph_a_mva": 2736433,
      "agg_s_ph_b_mva": 2498345,
      "agg_s_ph_c_mva": 2386196
    },
... ....
  "counters": {
    "main_CfgLoad": 1,
    "main_CfgChanged": 1,
    "main_taskUpdate": 8,
    "MqttClient_publish": 900,
    "MqttClient_respond": 18,
    "MqttClient_msgarrvd": 9,
    "MqttClient_create": 29,
    "MqttClient_setCallbacks": 29,
    "MqttClient_connect": 29,
    "MqttClient_connect_err": 15,
    "MqttClient_connect_Err": 15,
    "MqttClient_subscribe": 14,
    "SSL_Keys_Create": 29,
    "sc_SendStreamCtrl": 4,
    "sc_SendDemandRspCtrl": 1,
    "rest_Status": 48526
  },

Thanks in advance.

loadedtech, 2025-07-20:
Just noticed one other clue in the logs:
[homebridge-enphase-envoy] Device: 1 172.20.x.y Enphase, External integration start error: ReferenceError: client is not defined.

loadedtech, 2025-07-20:
how do I generate a "debug log file"?

grzegorz914, 2025-07-20:
[image]

loadedtech, 2025-07-20:
Here you go ... both the homebridge and plugin logs:
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21333390/homebridge-enphase-envoy.log.txt)
[homebridge.log.txt](https://github.com/user-attachments/files/21333389/homebridge.log.txt)

closed 2025-10-03 by grzegorz914