grzegorz914/homebridge-enphase-envoy #210: REST API Path EnsembleInventory not reachable

InRush79, 2025-07-26:
Hi!
I use the REST API integration for publishing the Data to other systems.
I just updated from an older 7 Version to the current 10 because the "consumptionToday" was broken (still seem with Version 10, it is stuck to always the same value)
The only way of reading the SoC of the EnCharge Battery was via REST API Path EnsembleInventory
this path shows only "This data is not available in your system."

[26.7.2025, 22:14:04] [homebridge-enphase-envoy] -------- Envoy --------
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Manufacturer: Enphase
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Model: ENV-S-WM-230
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Firmware: D8.3.5167
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] SerialNr: xxxxxxxxxxxx
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Time: 26.7.2025, 22:14:39
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] ------------------------------
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Q-Relays: 2
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Inverters: 14
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Wireless Kit: 2
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] --------------------------------
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Meters: Yes
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Production: Enabled
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Consumption: Enabled
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] --------------------------------
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Ensemble: Yes
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] Batt: 1
[26.7.2025, 22:14:04] [homebridge-enphase-envoy] --------------------------------

So ensemble should be activated automatically, but the REST API is not working as expected.

grzegorz914, 2025-07-27:
The path is `ensembleinventory`, please try also latest beta. If will not work post the debug please.

InRush79, 2025-07-27:
Thanks for your answer.
With the current beta it is running as expected.

The value of the path productionct->consumption->whToday seems to be wrong. shouldn't this be the sum of all three single lines? In my case my lines are at this moment 1390 / 2455 / 927 which is a sum of 4772 and makes sense. the value of whToday is 64697, a lot too high and makes no sense

InRush79, 2025-07-27:
Yes, I thought the same, because this behavior started without update of this plugin

grzegorz914, 2025-07-27:
Update to latest beta and use path`powerandenergydata`, I have added workarounds for the firmware bug. Let me know if is correct

grzegorz914, 2025-07-27:
Net and total are different things, should not be the same

grzegorz914, 2025-07-28:
Can you post the debug log, I will look into this. The values should be the same, I just summarized all lines.

InRush79, 2025-07-28:
[homebridge-enphase-envoy.log.txt](https://github.com/user-attachments/files/21470978/homebridge-enphase-envoy.log.txt)

Here you go, thank you for the quick responses. I also updated to latest beta.
I did this same workaround at my side to display the values.

closed 2025-09-05 by github-actions[bot]